chipsalliance/yosys-f4pga-plugins #8: Assertion violation in XDC plugin

litghost, 2020-02-19:
The symbiflow-arch-defs PR https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1321 causes the following assertion failure:
```
#terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
```

when the target `file_xc7_tests_ddr_ddr_uart_arty_artix7-xc7a50t-virt-xc7a50t-test_top.eblif` is built.  It is unclear if this is a latent yosys bug or a latent XDC plugin bug.

@tmichalak: Ideas?

litghost, 2020-02-19:
Regardless if this was a bug in the input data, the error message from yosys + the XDC plugin is not good enough.

litghost, 2020-02-19:
Small replication: [yosys_plugin.tar.gz](https://github.com/SymbiFlow/yosys-symbiflow-plugins/files/4227266/yosys_plugin.tar.gz)

tmichalak, 2020-02-19:
It seems to be related to some changes in Yosys during mapping as it works fine with `Yosys 0.9+932 (git sha1 81876a37, clang 8.0.0 -fPIC -Os)`, but fails with the assertion with `Yosys 0.9+1706 (git sha1 c244b27b, clang 6.0.0-1ubuntu2 -fPIC -Os)`. I will look into it.

litghost, 2020-02-21:
Any progress on this?

litghost, 2020-02-21:
I've identified the bug

litghost, 2020-02-25:
Fixed by #9

closed 2020-02-25 by litghost