chipsalliance/yosys-f4pga-plugins #13: XDC plugin doesn't understand port indexes properly

mkurc-ant, 2020-05-07:
It's a TCL thing. When a port is retrieved using eg. `[get_ports A[10]]` the TCL parser treats the port index in square brackets as another TCL command and fails there. This prevents us from using stock XDC files. A workaround for that is to specify the port like `[get_ports {A[10]}]`. 

A fix may be doing some kind of pre-processing of XDCs prior to feeding them to the Yosys' TCL interpreter.

mithro, 2020-05-07:
See @whitequark's twitter for how Vivado solves this problem.

mithro, 2020-05-07:
https://twitter.com/whitequark/status/1256506061462409218?s=20

mithro, 2020-05-07:
> i discovered something disturbing, but you have to know Tcl to understand it.
Vivado uses Tcl for scripting FPGA flows. Vivado allows using bracketed numbers in Tcl strings, e.g. [get_ports foo[1]] works. how?
> they made every integer a command that returns itself but bracketed

whitequark, 2020-05-08:
You can override [the [unknown] command](https://www.tcl.tk/man/tcl8.6/TclCmd/unknown.htm) to achieve this result.

mkurc-ant, 2020-05-08:
Thanks, that's interesting and definitely worth a shot.

PR #14 (2020-05-26): XDC: Fix port index parsing in get_ports command
This PR fixes #13 by adding a TCL rename proc during `get_ports` command registration.
closed 2020-05-26 by litghost