chipsalliance/yosys-f4pga-plugins #21: Build system does not respect `DESTDIR`

FFY00, 2020-06-26:
```
$ make PREFIX=/usr DESTDIR=dest install
make -C fasm-plugin test
make -C xdc-plugin test
make[1]: Entering directory '/home/anubis/git/symbiflow-arch-pkgs/yosys-symbiflow-plugins-git/src/yosys-symbiflow-plugins/fasm-plugin'
make[1]: Entering directory '/home/anubis/git/symbiflow-arch-pkgs/yosys-symbiflow-plugins-git/src/yosys-symbiflow-plugins/xdc-plugin'
gcc -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -Wall -Wextra -ggdb -I/usr/share/yosys/include -MD -D_YOSYS_ -fPIC -I/usr/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER -D_FORTIFY_SOURCE=2  -c -o xdc.o xdc.cc
gcc -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -Wall -Wextra -ggdb -I/usr/share/yosys/include -MD -D_YOSYS_ -fPIC -I/usr/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER -D_FORTIFY_SOURCE=2  -c -o fasm.o fasm.cc
mkdir -p /usr/share/yosys/plugins/fasm_extra_modules/
mkdir: cannot create directory ‘/usr/share/yosys/plugins’: Permission denied
make[1]: *** [Makefile:18: install_modules] Error 1
make[1]: *** Waiting for unfinished jobs....
```

Xiretza, 2021-01-22:
Should be as easy as adding a `$(DESTDIR)` to https://github.com/SymbiFlow/yosys-symbiflow-plugins/blob/2894db22d21c055606612be35b78fe87746dc161/Makefile_plugin.common#L43

tcal-x, 2022-04-19:
I came here to file the same issue.

https://www.gnu.org/prep/standards/html_node/DESTDIR.html

mkurc-ant, 2022-04-21:
The thing with Yosys plugins is that they have to be installed in a specific directory in order for Yosys to find them later. This happens automatically because the build system relies on Yosys binaries (namely `yosys-config`). Why do you need to insall them to a different directory?

If you want to build and install plugins for Yosys that resides under a specific path you may set the `YOSYS_PATH` variable accordingly while invoking `make`
https://github.com/chipsalliance/yosys-f4pga-plugins/blob/55c26d7629a3728a6527e238bd2b77b7740150d1/Makefile_plugin.common#L42

Xiretza, 2022-04-21:
> Why do you need to insall them to a different directory?

In order to create a package. All files are `make install`ed to a staging directory (`$DESTDIR`), which is then `tar`ed up by the packaging scripts.

> If you want to build and install plugins for Yosys that resides under a specific path you may set the `YOSYS_PATH` variable accordingly while invoking `make`

No, `yosys` and `yosys-config` are installed in their proper place (`/usr/bin/`) in the build environment, nothing needs to be changed there.

alanvgreen, 2022-04-21:
@mkurc-ant It is possible to build plugins with yosys at a non-default location. For example, if yosys is to be installed in `/foo/bin`, first build yosys with `make PREFIX=/foo`. This creates a `yosys-config` with the correct paths (i.e to /foo/bin, /foo/share etc). 

Then, as recommended by @Xiretza, build yosys plugins, with `YOSYS_PATH=/foo make`.

See https://github.com/google/CFU-Playground/blob/main/scripts/install_oxide#L73

mkurc-ant, 2022-04-22:
@alanvgreen Yes, that's definitely possible but as I understand @Xiretza wants to override those paths anyways. Let me work on adding that to the makefile.

mkurc-ant, 2022-04-22:
I've created a PR which enables specifying `DESTDIR`: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/306

Xiretza, 2022-06-01:
Later amended by #323.