chipsalliance/yosys-f4pga-plugins #30: Split SDC plugin into separate timing constraints related plugins

tmichalak, 2020-09-02:
The objective is to divide the functionalities of the current SDC plugin into 2 optionally 3 separate plugins:
1. 'sdc' -- The part which reads SDC into the Yosys attributes, i.e. `read_sdc` and `write_sdc`
2. 'propagate-constraints' -- The part which propagates the constraints in the Yosys attributes across things like clock buffers and similar, i.e. `propagate_clocks` (and maybe future things like generate_pll_config or something?)
3. (optionally) 'pcf' -- In case we end up wanting a `read_pcf` or `read_other_timing_constraint_format` plugins which pull data into the same Yosys attributes.

In the current SDC plugin implementation the clock information is not assigned to parameters but kept in a separate `Clocks` class object. The object would have to be global in order to work for all timing related plugins or the clock information would have to be stored in the RTLIL::Wire parameters in form of extra parameters.

tmichalak, 2020-09-02:
@mithro @litghost Please have a look if the description correctly captures the main idea of the SDC plugin split.

mithro, 2020-09-02:
This is correct.

I would say that storing the information in parameters is a separate issue that should be logged.