chipsalliance/yosys-f4pga-plugins #33: Add missing SDC commands used in Litex generated XDC

tmichalak, 2020-09-07:
The SDC commands used in the auto-generated XDC [PR](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1657) are:

- create_clock
- set_false_path
- set_max_delay 

`create_clock` is already supported in the [XDC plugin] (https://github.com/SymbiFlow/yosys-symbiflow-plugins/pull/27), however the target net is wrapped in the `get_nets` command which needs to be handled.

In the first approach all the `set_false_path` and `set_max_delay` commands shall simply be forwarded to the output SDC and passed 'as is' to VPR.

mithro, 2020-09-09:
@tmichalak - The `set_false_path` and `set_max_delay` should be applied to the correct cells and then the data extracted again when writing out the SDC. This is important as the cells might change names, be removed, etc due to optimizations and stuff.

tmichalak, 2020-09-09:
@mithro Yes I figured out that simply writing out these commands as the come in in the original SDC will not work. 
For example for minilitex_ddr we have:
```
set_false_path -quiet -to [get_pins -quiet -filter {REF_PIN_NAME == PRE} -of [get_cells -quiet -filter {ars_ff1 == TRUE || ars_ff2 == TRUE}]]

set_max_delay 2 -quiet -from [get_pins -quiet -filter {REF_PIN_NAME == Q} -of [get_cells -quiet -filter {ars_ff1 == TRUE}]] -to [get_pins -quiet -filter {REF_PIN_NAME == D} -of [get_cells -quiet -filter {ars_ff2 == TRUE}]]
```
so this looks more like applying it to the correct pin. 
However, just by looking at the above example it seems that we will need to have the `get_pins` command support filtering and stuff just like Vivado. 
For the first version of the SDC plugin we could:
1. Ignore/filter out these commands at all
2. Support only commands with explicit names of the pins like `set_max_delay 2 -from pin1 -to pin2`

Option 1 is already supported as the read_xdc command overrides the TCL specific `unknown` procedure handler which results in simply ignoring unknown commands (so in this case all missing SDC commands).

However my guess is that we are leaning towards option 2, right?

mithro, 2020-09-09:
I think we want to support the filtering commands? They should be pretty simple right?

tmichalak, 2020-12-15:
Basic versions of all of the necessary commands have been implemented.

closed 2020-12-15 by tmichalak