chipsalliance/yosys-f4pga-plugins #39: read_xdc command accepts unknown commands

tmichalak, 2020-09-22:
Currently if the file that is read with the read_xdc command contains an unknown command no error is reported.
This is caused by the fact that read_xdc makes use of the `unknown args` procedure and implements the handler which returns the string of the called command surrounded by square bracket.
We need to make sure commands that are not number will error out.

litghost, 2020-09-22:
> We need to make sure commands that are not number will error out.

Also if a number if supplied with more arguments.

So examples:
```
[0] -> [0]
[0 1] -> error
[get_ports 0] -> error
[get_ports] -> error
```

litghost, 2020-09-22:
Also we discussed having an option on `read_xdc` to control whether the behavior is a warning or an error.