chipsalliance/yosys-f4pga-plugins #45: Some plugins hold live pointers beyond liveness scope

litghost, 2020-10-08:
The current implementation of the SDC plugin holds RTLIL objects between TCL commands.  Because of this, changes to those pointers (e.g. techmap passes, etc) cause those pointers to become invalid.

Test case for the SDC plugin attached: [test.zip](https://github.com/SymbiFlow/yosys-symbiflow-plugins/files/5351665/test.zip)

litghost, 2020-10-08:
There may be more examples of this.  In general, passes can only ensure liveness of RTLIL objects within the scope of the pass, and no longer.  Holding these pointers between invocations is not safe, and will result in memory corruption.

litghost, 2020-10-08:
@tmichalak This is the highest priority issue, and must be resolved.  This is not optional.

tmichalak, 2020-10-09:
@litghost so for SDC the issue is that we are storing pointers to RTLIL::Wire* in the Clock objects, right?
In case Yosys destroys the wires what should be the desired behavior apart from not crashing (in this a simple !=NULL will do)?
Removing the Wire and clock if that is the last or only one clock wire?

litghost, 2020-10-09:
> @litghost so for SDC the issue is that we are storing pointers to RTLIL::Wire* in the Clock objects, right?
> In case Yosys destroys the wires what should be the desired behavior apart from not [...]

You cannot hold the pointers at all.  It is not a safe behavior.  My recommendation is to defer module inspection until when you are ready to do so (e.g. write_sdc).  Having intermediate `RTLIL::Wire*` objects between TCL calls/yosys passes will always be unsafe.

Pointer checks are useless here, because yosys deleting an address won't update the pointer copy held in the Clocks object.

tmichalak, 2020-10-19:
@litghost I created a draft PR targeting the issue.
Basically, I created a write_sdc script pass command which wraps the `propagate_clocks` and `write_sdc` commands effectively postponing the clock propagation to when `write_sdc` is called.
As for the pointers, I am just cleaning the clock structure after each run, so this is not optimal since every `write_sdc` triggers a new clock propagation step.
I am aware that we can't keep pointers between runs, so maybe instead of keeping `RTLIL::Wire*` objects, I could just keep the wire name. 
However, in the test you added to the issue I think that cleaning the clock structure is actually the desired behavior.
One thing I noticed is that the second pass in your test just reads the design from json that was written in the first pass.
Currently that json doesn't keep any clock information, it probably should but that would require enhancing the `write_json` command. For know, you have to add the clocks (call all SDC commands again).

litghost, 2020-10-19:
> For know, you have to add the clocks (call all SDC commands again).

This is correct, but the SDC plugin doesn't get a signal from Yosys that the relevant objects no longer exists, hence the memory corruption angle.

PR #51 (2020-10-29): Keep the SDC information in Yosys's design structures
This PR fixes #45 by modifying the current SDC plugin implementation to not keep the clock and generated clock information in a separate structure, but in Yosys' internal design representation. For this purpose the clock related attributes from [this list](https://docs.google.com/spreadsheets/d/1G-E2Dq8YG4g9Z6mTygpumwlI_vNlFUQinc9gMgePfec/edit#gid=1349094793) on the specified wires are set or updated using `create_clocks` or as a result of clock propagation.
closed 2020-10-29 by litghost