chipsalliance/yosys-f4pga-plugins #59: Remove workaround with skipping clocks on dangling wires

tmichalak, 2020-12-02:
PR https://github.com/SymbiFlow/yosys-symbiflow-plugins/pull/57 has been created to fix the [reported problem](https://github.com/SymbiFlow/symbiflow-arch-defs/issues/1753#issuecomment-733165240).
This is needed to the 200T designs to work with SymbiFlow, but the problem is in fact in VPR  as it sweeps the netlist and invalidates an SDC that was originally valid.

mithro, 2020-12-02:
@tmichalak - Didn't we discover that there are two issues;
 - (a) Yosys is producing output in the BLIF that VPR sweeps.
 - (b) VPR is sweeping away stuff but not updating the SDC.

tmichalak, 2020-12-02:
>  @tmichalak - Didn't we discover that there are two issues;
>
> (a) Yosys is producing output in the BLIF that VPR sweeps.
> (b) VPR is sweeping away stuff but not updating the SDC.

@mithro Yes, that is correct.