chipsalliance/yosys-f4pga-plugins #62: Fix curly brackets support in get_ports commands

rw1nkler, 2020-12-08:
Many public XDC files provide `get_ports` arguments in curly brackets:
https://github.com/Digilent/digilent-xdc/blob/master/Arty-A7-35-Master.xdc#L7

Currently, support for this feature is missing / incomplete. The following XDC file:
```
set_property PACKAGE_PIN E3 [get_ports { clk }]
set_property IOSTANDARD LVCMOS33 [get_ports { clk }]
set_property PACKAGE_PIN H5 [get_ports { led }]
set_property IOSTANDARD LVCMOS33 [get_ports { led }]
set_property PACKAGE_PIN A8 [get_ports { sw }]
set_property IOSTANDARD LVCMOS33 [get_ports { sw }]
create_clock -period 10.0 clk
```
causes the following error during the Yosys synthesis step:
```
...
7.38.3. Continuing TECHMAP pass.
Using template $paramod\$lut\WIDTH=1\LUT=2'01 for cells of type $lut.
No more expansions possible.
7.39. Executing XILINX_DFFOPT pass (optimize FF control signal usage).
Optimizing FFs in top.
7.40. Executing OPT_LUT_INS pass (discard unused LUT inputs).
Optimizing LUTs in top.
Removed 0 unused cells and 2 unused wires.
8. Executing HIERARCHY pass (managing design hierarchy).
8.1. Analyzing design hierarchy..
Top module:  \top
8.2. Analyzing design hierarchy..
Top module:  \top
Removed 0 unused modules.
set_property PACKAGE_PIN E3 [get_ports { clk }]
set_property IOSTANDARD LVCMOS33 [get_ports { clk }]
set_property PACKAGE_PIN H5 [get_ports { led }]
set_property IOSTANDARD LVCMOS33 [get_ports { led }]
set_property PACKAGE_PIN A8 [get_ports { sw }]
set_property IOSTANDARD LVCMOS33 [get_ports { sw }]
create_clock -period 10.0 clk
Warning: Couldn't find port matching  clk
ERROR: set_property IO_LOC_PAIRS: Incorrect number of arguments.
```

After removing the curly brackets, everything works as intended:
```
set_property PACKAGE_PIN E3 [get_ports clk]
set_property IOSTANDARD LVCMOS33 [get_ports clk]
set_property PACKAGE_PIN H5 [get_ports led]
set_property IOSTANDARD LVCMOS33 [get_ports led]
set_property PACKAGE_PIN A8 [get_ports sw]
set_property IOSTANDARD LVCMOS33 [get_ports sw]
create_clock -period 10.0 clk
```

IshwarJeet, 2020-12-08:
I'm new to pragrammimg. let me handle this issue

rw1nkler, 2020-12-08:
Sure! In case you need any additional information, don't hesitate to ask! 
Me and @tmichalak will be happy to help you

IshwarJeet, 2020-12-08:
can you assist me in this

rw1nkler, 2020-12-08:
Of course! Here is the plugin source file:
https://github.com/SymbiFlow/yosys-symbiflow-plugins/blob/master/xdc-plugin/xdc.cc

First, make sure that you can compile Yosys and run it with the XDC plugin.
This requires setting some options for Yosys, or copying the files manually. 

Tomorrow, I can give you more detailed information how to do that and test the plugin.

IshwarJeet, 2020-12-12:
tell me more on how to do that

rw1nkler, 2020-12-12:
Here are instructions that will allow you to reproduce my error. I created another branch with the failing syntax, so it will be easier for you to set up a working project.

1. Create a workspace directory:

```
mkdir yosys_plugins_workspace
cd yosys_plugins_workspace
```

2. Download, compile, and install `yosys`. It is also possible to proceed without installing `yosys` globally but this option requires a more complicated `yosys-plugins` setup.
```
git clone https://github.com/symbiflow/yosys
cd yosys
make j$(nproc)
sudo make install
cd ..
```

3. Download, compile, and install `yosys-plugins`:
```
git clone https://github.com/SymbiFlow/yosys-symbiflow-plugins
cd yosys-symbiflow-plugins
make j$(nproc)
sudo make install
cd ..
```

4. Download the `symbiflow-arch-defs` repository which contains tests of the symbiflow toolchain and prepare the local conda environment:
```
git clone --recursive https://github.com/antmicro/symbiflow-arch-defs -b bufgmux_broken
cd symbiflow-arch-defs
export YOSYS=`which yosys`
make env
source env/conda/bin/activate symbiflow_arch_def_base
```

`yosys` + `yosys_plugins` are used in the toolchain synthesis step. Because of that, you can use the examples provided there to check whether your changes work correctly with the prepared synthesis script. You can use your system `yosys` installation after you set the `YOSYS` environment variable. The variable has to point to the `yosys` binary in your system. 

5. Run the example to reproduce my errors:
```
cd build/xc/xc7/tests/bufgmux
make bufgmux_arty_bit
```

Note that this may take about 1h to generate the FPGA architecture for the chip. 
On the other hand, you will be able to use other examples to test your changes.

Make sure that after changing the XDC plugin in the `yosys-plugin` repository, you built and installed the latest version of your plugin.

PR #67 (2020-12-17): get_ports: Trim port name
This PR fixes #62
closed 2020-12-17 by litghost