chipsalliance/yosys-f4pga-plugins #74: macOS: There's no `-D` switch in BSD `install`

ajelinski, 2021-01-26:
One of the latest commits (https://github.com/SymbiFlow/yosys-symbiflow-plugins/commit/c613a356f3b5a6e93d2ba64f3568f4e47cef4768) added the `install -D` command.

After that change building the *plugins* on macOS fails in the https://github.com/litex-hub/litex-conda-eda repository.

The BSD's `install -d` seems to have a similar meaning to the GNU's `install -D` but I'm not sure about that:
```
     -d      Create directories.  Missing parent directories are created as required.
```

It'd be great if the script could either have `install` used universally to both implementations or simply have the commands replaced in the commit with `install` as a fail-over for macOS.

hzeller, 2021-03-22:
Oh, sorry, just seen this bug assignment. Will make something that should work in both.

timkpaine, 2023-04-14:
I'm working on this FYI. I have 100% working Mac build locally, just trying to deal with propagating changes through all the deps. My changes remove all the conda stuff (separately I'm moving surelog / UHDM onto conda-forge, I'd like to obviate the need for a separate Litex-hub channel ideally).

timkpaine, 2023-04-14:
@hzeller I'm not a member of this repo I don't think but feel free to assign off to me