chipsalliance/yosys-f4pga-plugins #78: How to make get_count work in Tcl scripts?

donn, 2021-02-03:
I'm using the version hosted in litex-hub's conda repo thing.

I must say, I'm a bit mystified by something. In that version, with the environment set up using conda, tcl scripts using get_count work fine.

But when I set it up manually by compiling both yosys and the plugins, I get this error:

```
ERROR: TCL interpreter returned an error: invalid command name "get_count"
```

Curiously, the xdc plugin works fine even when compiled from scratch.  I thought I might be using the wrong version for get_count, so I checked the shared objects on the conda installation and well, it appears that the symbol is only there in xdc.so still.

```
(eos-s3) [root@334715b73679 plugins]# nm get_count.so  | grep -i tcl
                 U Tcl_Eval
                 U Tcl_ListObjAppendElement
                 U Tcl_NewListObj
                 U Tcl_NewStringObj
                 U Tcl_SetObjResult
                 U _ZN5Yosys20yosys_get_tcl_interpEv
(eos-s3) [root@334715b73679 plugins]# nm xdc.so  | grep -i tcl
                 U Tcl_Eval
                 U Tcl_EvalFile
                 U Tcl_GetStringResult
                 U Tcl_SetResult
00000000000060c0 t _ZN12_GLOBAL__N_127register_in_tcl_interpreterERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
                 U _ZN5Yosys20yosys_get_tcl_interpEv
```                 

So uh, my question is, how does get_count work in the Tcl scripts for conda without registering itself?

Xiretza, 2021-02-03:
What is the exact yosys command line you're using? The following works for me:

```
$ cat test.tcl
yosys plugin -i get_count

yosys -import

puts [get_count -modules]
$ yosys test.tcl
[...]
 Yosys 0.9+3882 (git sha1 beeaad190, gcc 10.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fvar-tracking-assignments -fdebug-prefix-map=/build/yosys-git/src=/usr/src/debug -fPIC -Os)

-- Parsing `test.tcl' using frontend `tcl' --
[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
0
```

donn, 2021-02-03:
Just the standard eos-s3 ql_symbiflow script. 

```
[root@d1673f00302e btn_counter]# PATH=/symbiflow-docker/ql_eos-s3/install/bin/:$PATH make
ql_symbiflow -compile -d ql-eos-s3 -P pd64 -v btn_counter.v -t top -p chandalar.pcf
----------------- 

-compile
-d
ql-eos-s3
-P
pd64
-v
btn_counter.v
-t
top
-p
chandalar.pcf
verilog files: btn_counter.v 
Running Synth->Pack->Place->Route 
make[1]: Entering directory `/symbiflow-examples/eos-s3/btn_counter'
cd build && synth -t top -v /symbiflow-examples/eos-s3/btn_counter/btn_counter.v   -d ql-eos-s3_wlcsp -p /symbiflow-examples/eos-s3/btn_counter/chandalar.pcf -P PD64 2>&1 > /symbiflow-examples/eos-s3/btn_counter/build/top.log
ERROR: TCL interpreter returned an error: invalid command name "get_count"
make[1]: *** [build/top.eblif] Error 1
make[1]: Leaving directory `/symbiflow-examples/eos-s3/btn_counter'
make: *** [all] Error 2
```

donn, 2021-02-03:
I should mention your minimal script worked fine so I have absolutely no clue what's going on anymore.

Xiretza, 2021-02-03:
Your `ql_symbiflow` still uses the `synth` script without `symbiflow_` prefix, which was changed a few months ago [here](https://github.com/QuickLogic-Corp/symbiflow-arch-defs/commit/dddcbc4b33d9785720dc0134e8062e0dc350f762#diff-d42600a744c15753bb444867d01ec9bad97a36998f1b3b05997214e58141dae4R384), so looking for a more recent version might help. Other than that I can't really help though, I've never used any of these docker containers or conda environments.

donn, 2021-02-03:
Thanks for the help!

tmichalak, 2021-02-03:
@donn I believe you question has been answered? And issue can be closed?

donn, 2021-02-03:
I'm trying the new ql_symbiflow and will report accordingly.

donn, 2021-02-03:
No dice, I'm afraid. Thanks for the help regardless.

closed 2021-02-03 by donn