chipsalliance/yosys-f4pga-plugins #144: Wrong synthesis result when buses are not 0-addressed

gergoerdi, 2021-11-12:
This is in relation to https://github.com/YosysHQ/yosys/issues/3078 where I was told that the debug output of Yosys is just misleading, i.e. the problem should be elsewhere in the toolchain.

Given the following Verilog input:

```
module Top(
           output [5:2] LED
           );

   assign LED[5:2] = 4'b1101;
endmodule
```

I am synthesizing for the Nexys A7 50T dev board with a Xilinx xc7a50tcsg324 FPGA. I connect `LED[5:2]` to the LEDs on board with the following XDC:

```
set_property -dict { PACKAGE_PIN J13   IOSTANDARD LVCMOS33 } [get_ports { LED[2] }];
set_property -dict { PACKAGE_PIN N14   IOSTANDARD LVCMOS33 } [get_ports { LED[3] }];
set_property -dict { PACKAGE_PIN R18   IOSTANDARD LVCMOS33 } [get_ports { LED[4] }];
set_property -dict { PACKAGE_PIN V17   IOSTANDARD LVCMOS33 } [get_ports { LED[5] }];
```

If I synthesize this with Vivado, LEDs 2, 4 and 5 light up, as expected. If I synthesize this with Symbiflow, instead only LED 2 lights up. 

Full reproduer is at https://github.com/gergoerdi/yosys-issue-3078/tree/81f4cd6878574614f08269e44ec8df72a857c640

mithro, 2021-11-12:
I'm pretty sure this is an issue with the XDC plugin as a number of the `OBUFT_VPR` modules do not have `IO_LOC_PAIRS` value set.

mithro, 2021-11-12:
PTAL - @tmichalak

tmichalak, 2021-11-15:
Yes, this looks like a bug in the plugin. I will look into it.

tmichalak, 2021-12-13:
Closed with #161

closed 2021-12-13 by tmichalak