chipsalliance/yosys-f4pga-plugins #148: Building uhdm plugin fails due to missing uhdm.h

olofk, 2021-11-17:
```
uhdmastreport.h:9:10: fatal error: uhdm/uhdm.h: No such file or directory
    9 | #include <uhdm/uhdm.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [../Makefile_plugin.common:63: UhdmAst.o] Error 1
make[1]: Leaving directory '/home/olof/projects/edalize/yosys-symbiflow-plugins/uhdm-plugin'
make: *** [Makefile:37: uhdm.so] Error 2
```

kgugala, 2021-11-18:
@olofk I believe this is already solved. Can we close this issue?

olofk, 2021-11-18:
Yes. It could be worth mentioning where to find the uhdm library, but I understand that things are still up in the air and that https://github.com/antmicro/yosys-uhdm-plugin-integration is the best entrypoint for now

closed 2021-11-18 by olofk