chipsalliance/yosys-f4pga-plugins #177: ERROR: Incorrect period value

fontamsoc, 2021-12-29:
https://lists.librecores.org/pipermail/symbiflow/2021-December/000049.html

Hello,

I am attempting to build [FonTamSOC](https://github.com/fontamsoc)
targeting NexysA7.

However, `symbiflow_synth` fails with `ERROR: Incorrect period value`
at the end of the logs.

Steps to reproduce failure:
```
# clone verilog sources:
git clone https://github.com/fontamsoc/hw.git
cd hw

# prerequisite:
ln -snf pu32-nexys4ddr/litedram/litedram.hex litedram.hex

# synthesize:
symbiflow_synth -t nexys4ddr -v pu32-nexys4ddr/nexys4ddr.v -d artix7
-p xc7a100tcsg324-1 -x pu32-nexys4ddr/nexys4ddr.xdc &>/dev/null

tail nexys4ddr_synth.log
```

> The error seems to come from
> https://github.com/SymbiFlow/yosys-symbiflow-plugins/blob/master/sdc-plugin/sdc.cc#L153
> 
> I'm guessing that the error is coming from line 4 of your .xdc file but the
> current output doesn't really give you enough information to know or
> confirm that. I recommend logging a bug on either symbiflow-examples or
> yosys-symbiflow-plugins with your reproduction instructions and output you
> have provided.
> 
> Next steps would probably be to improve the error message with the value it
> thinks is invalid. The code also doesn't seem to know the difference
> between the create_clock command getting invalid value for -period and the
> command not getting a -period argument all together.
> 
> Second step would be to figure out how to create / throw the error message
> so it reports the file and line number where the issue is found.
> 
> Hope that helps,
> 
> Tim 'mithro' Ansell

tmichalak, 2021-12-29:
@fontamsoc I couldn't reproduce the error on my end. What version of the symbiflow-plugins do you use? Is it the same as [here](https://github.com/SymbiFlow/symbiflow-examples/blob/master/xc7/environment.yml#L11) ?
Have you ever built the sdc plugin manually. If so then there is a [debug option](https://github.com/SymbiFlow/yosys-symbiflow-plugins/blob/master/Makefile_plugin.common#L51) that prints out additional information during the clock propagation phase.
Also, should I use the code on the master branch or some other commit, because the `44c4298` SHA that is in the report can't be found.

fontamsoc, 2021-12-29:
HI @tmichalak,

I have updated steps to reproduce to `git checkout issues/177` which is the same SHA `44c4298`.

I have never built sdc plugin manually; however I have installed SymbiFlow using following instructions:
https://symbiflow-examples.readthedocs.io/en/latest/getting-symbiflow.html

I was able to work around the issue removing `-add` from `create_clock` at:
https://github.com/fontamsoc/hw/blob/44c4298/pu32-nexys4ddr/nexys4ddr.xdc#L4

mithro, 2021-12-29:
FYI - I logged the following GitHub issues based on this bug report;
 - https://github.com/SymbiFlow/yosys-symbiflow-plugins/issues/179
 - https://github.com/SymbiFlow/yosys-symbiflow-plugins/issues/178
 - https://github.com/SymbiFlow/yosys-symbiflow-plugins/issues/180

olofk, 2022-01-26:
Yep. Hitting the same bug here as described in https://github.com/SymbiFlow/yosys-symbiflow-plugins/issues/180 . Removing `-add` makes it pass