chipsalliance/yosys-f4pga-plugins #218: UHDM Plugin: Anonymous structs fail to be converted to AST in current plugin

QuantamHD, 2022-02-08:
Example module

```systemverilog
module anon_module 

   (
    output logic                                    all_done 
    );

   typedef struct packed {
     logic                      a;
     logic                      b;
   } typedef_struct_thing;

   typedef_struct_thing [50:0]   data;  

   struct packed {
     logic a;
     logic [100:0]      b;
   } [200:0] anon_struct;
     

   always_comb begin
     data[0].a = 0; //ok
     anon_struct[0].a = 0; // Not ok
     all_done = anon_struct.a;
   end

endmodule : anon_module
```

This example will fail to be read into yosys. Specifically it lacks the information added by add_multirange_attribute both multirange inputs are totally empty in the case of anon_struct, but are filled in for data. Giving a typedef name to the anon struct also fixes the issue.

QuantamHD, 2022-02-11:
@kbieganski @alaindargelas added more details in the linked issue above.

QuantamHD, 2022-02-14:
@hzeller Can you help prioritize this issue. It's one of the blockers on the internal project

kamilrakoczy, 2022-02-15:
@QuantamHD It looks like now we have all necessary information in UHDM file. I've started implementing support for this in uhdm-plugin.

kamilrakoczy, 2022-02-16:
@QuantamHD I merged PR adding support for this to master.

closed 2022-02-16 by kamilrakoczy