chipsalliance/yosys-f4pga-plugins #249: dsp_ff vs dsp-ff

tcal-x, 2022-02-24:
@mkurc-ant  -- this may be a packaging issue -- I'm using `yosys` and `yosys-symbiflow-plugins` packages from litex-hub -- I get an error from Yosys:
```
ERROR: Can't load module `./dsp_ff': /media/tim/GIT/google/CFU-Playground/env/conda/envs/cfu-common/bin/../share/yosys/plugins/dsp_ff.so: cannot open shared object file: No such file or directory
```
but there is a file
```
/media/tim/GIT/google/CFU-Playground/env/conda/envs/cfu-common/bin/../share/yosys/plugins/dsp-ff.so
```

(note `dsp_ff.so` vs `dsp-ff.so`)

mkurc-ant, 2022-02-24:
@tcal-x As of yesterday the conda package containing the plugin should be present with the `main` label. Now I can see that I made a typo in the PR description, the plugin file is named `dsp-ff.so` but the Yosys pass name is `dsp_ff`. So you should do `plugin -i dsp-ff` but then `dsp_ff` to use it.

tcal-x, 2022-02-24:
Thanks @mkurc-ant that does resolve the issue for me.

closed 2022-02-24 by tcal-x