chipsalliance/yosys-f4pga-plugins #253: Is local install supported?

tcal-x, 2022-02-25:
I don't see any documentation on how to perform a local install (i.e. under `/usr/local`).

If I run `make`, most of the sub-builds succeed generating `.so` files, although I get this error (@hzeller ?):
```
In file included from uhdmastshared.h:4,
                 from UhdmAst.h:8,
                 from uhdmsurelogastfrontend.cc:22:
uhdmastreport.h:9:10: fatal error: uhdm/uhdm.h: No such file or directory
    9 | #include <uhdm/uhdm.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
```
When I try `sudo make install`, I get
```
make -C fasm-plugin install
make[1]: Entering directory '/home/tcal/SymbiFlow/yosys-f4pga-plugins/fasm-plugin'
../Makefile_plugin.common:47: *** "Didn't find 'yosys-config' under '/'".  Stop.
make[1]: Leaving directory '/home/tcal/SymbiFlow/yosys-f4pga-plugins/fasm-plugin'
make: *** [Makefile:37: install_fasm] Error 2
```

This seems to be some strange interaction between `sudo` and the calculation of `YOSYS_PATH` in `Makefile_plugin.common`.   I do have both `/usr/local/bin/yosys` and `/usr/local/bin/yosys-config`, so `YOSYS_PATH` should be `/usr/local`.    At my command line, `sudo which yosys` returns nothing, even though `sudo echo $PATH` contains `/usr/local/bin`.  

A workaround is to run `sudo make install YOSYS_PATH=/usr/local`, but it seems something is broken.

hzeller, 2022-02-25:
https://github.com/antmicro/surelog-uhdm-ibex-guide gives a good overview what needs to be done.

Essentially, you need Surelog and UHDM present for the module to compile.

tcal-x, 2022-02-26:
> even though `sudo echo $PATH` contains `/usr/local/bin`.

Ah, this is wrong as pointed out in https://unix.stackexchange.com/questions/32764/why-which-tmux-and-sudo-which-tmux-return-2-different-values -- $PATH gets expanded before getting passed into `sudo`.   If I instead do:
```
sudo sh -c 'echo $PATH'
```
... I get:
```
/usr/sbin:/usr/bin:/sbin:/bin
```
..which doesn't include `/usr/local/bin`.

So, the mechanism for finding `yosys-config` doesn't work as implemented when run under `sudo`.