chipsalliance/yosys-f4pga-plugins #313: Local function inside generate statement can't be resolved

hzeller, 2022-05-06:
Consider the following; important is the local function inside the generate statement which then is used right after it is defined

```systemverilog
module C #(
    parameter int width = 2,
    parameter DO_INCREMENT = 1
) (
    input clk,
    output [width-1:0] counter
);

  logic [width-1:0] foo;
  generate
    if (DO_INCREMENT == 1) begin
      function automatic [width-1:0] local_increment_function(input [width-1:0] x);
        local_increment_function = x + 1;
      endfunction
      assign foo = local_increment_function(counter);
    end else begin
      assign foo = counter;
    end
  endgenerate
  // ... do stuff with foo
endmodule

module make_instances (
    input clk,
    output logic [31:0] foo
);
  C #(
      .DO_INCREMENT(1)
  ) c (
      clk,
      foo
  );
endmodule
```

Generating UHDM with Surelog and then reading it with yosys with the [systemverilog plugin](https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin) compiled in

```
surelog -nobuiltin -verbose -parse -sv filename-above.sv
yosys -p "read_uhdm ${SURELOG_TMP}/slpp_all/surelog.uhdm"
```

... will then complain that it can't resolve the function `\local_increment_function`
```
1. Executing UHDM frontend.
Warning: Removing unused module: \C from the design.
Generating RTLIL representation for module `$paramod\C\width=32'd2\DO_INCREMENT=32'd1'.
fun_in_generate.sv:15: ERROR: Can't resolve function name `\local_increment_function'.
```

Expectation is that the function can be resolved and used (processing the same file with 'plain' yosys works fine).

It is unclear if this is a Surelog -> UHDM name manglilng problem or a UHDM -> Yosys plugin problem. So adding here to triage.

CC @alaindargelas

alaindargelas, 2022-05-06:
I'll check the former possibility

alaindargelas, 2022-05-07:
The problem is in the UHDM->Yosys plugin.
Per Standard the function definition is attached to the module instance, not inside the generate statement, so the plugin has to either:
- lookup for the definition traversing the parent scope recursively (There could be many nested generate statements) until it reaches a module instance, then traverse the list of task_funcs
- Simply from the function call, get the pointer to the function definition.

Another solution, if it is a Yosys lookup issue is to inline the function definition in the local scope for Yosys (Replicate the function def at the module level and at the generate stmt level.

alaindargelas, 2022-05-07:
After checking several scenario with commercial simulator I concluded that the VPI Object model for gen_scope is wrong, I added a task_func property in the model and now this allows to have local declaration of tasks and funcs local to the gen_scope.
Prior to this UHDM was storing the function definition in the parent module of the gen_scope (Due to the limitation of the Standard Object Model).
There must be a corresponding change that needs to happen in the plugin to iterate through the newly added function declaration in the gen_scope.

https://github.com/chipsalliance/UHDM/pull/711
https://github.com/chipsalliance/Surelog/pull/2915

hzeller, 2022-05-07:
Thanks @alaindargelas 
@tgorochowik @rkapuscik Adding the corresponding relevant support in the plugin would be a high priority request from us.

alaindargelas, 2022-05-07:
Please also note that this slightly different case is legal (The hier path access into the gen_scope from outside the gen_scope part):
https://github.com/chipsalliance/Surelog/blob/master/tests/GenScopeFunc/dut.sv
And is now supported:
https://github.com/chipsalliance/Surelog/pull/2916

rkapuscik, 2022-05-09:
@hzeller #314 should make this work (your example passes for me now), please confirm.

closed 2022-05-09 by rkapuscik

hzeller, 2022-05-10:
affirmative :)

PR #314 (2022-05-09): Parse task/function definitions inside generate blocks
Fixes #313.