chipsalliance/yosys-f4pga-plugins #317: Identifier defined in a named block can not be resolved ('Failed to detect width for identifier ...')

hzeller, 2022-05-11:
A register, declared inside a named block can not be resolved (Yosys reports that it doesn't know the 'width' of the identifier, but that is probably due to it not resolving it properly).

Consider this test case:

```systemverilog
module top();
  always @* begin : named_block
    reg[0:0] foo;
    foo = 1;
  end
endmodule
```

Running this through surelog -> UHDM then reading it in Yosys results in:
```
1. Executing UHDM frontend.
Generating RTLIL representation for module `\top'.
width_unknown.v:4: ERROR: Failed to detect width for identifier \named_block.foo!
```

So it looks like, Yosys attempts to look up this name `foo` with the `named_block` namespace prefix and doesn't find it.

This works if the block is _not_ named (so, `: named_block` removed).

This is how the UHDM dump looks like
```
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) width_unknown.v:1:1: , endln:6:10, parent:work@top
  |vpiFullName:work@top
  |vpiDefName:work@top
  |vpiProcess:
  \_always: , line:2:3, endln:5:6, parent:work@top
    |vpiStmt:
    \_event_control: , line:2:10, endln:2:12
      |vpiStmt:
      \_named_begin: (work@top.named_block), line:2:13, endln:5:6
        |vpiName:named_block
        |vpiFullName:work@top.named_block
        |vpiVariables:
        \_logic_var: (work@top.named_block.foo), line:3:14, endln:3:17, parent:work@top.named_block
          |vpiName:foo
          |vpiFullName:work@top.named_block.foo
          |vpiRange:
          \_range: , line:3:9, endln:3:12, parent:work@top.named_block
            |vpiLeftRange:
            \_constant: , line:3:9, endln:3:10
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
              |vpiConstType:9
            |vpiRightRange:
            \_constant: , line:3:11, endln:3:12
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
              |vpiConstType:9
        |vpiVariables:
        \_logic_var: (work@top.named_block.foo), line:3:14, endln:3:17, parent:work@top.named_block
        |vpiStmt:
        \_assignment: , line:4:5, endln:4:12, parent:work@top.named_block
          |vpiOpType:82
          |vpiBlocking:1
          |vpiRhs:
          \_constant: , line:4:11, endln:4:12
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
            |vpiConstType:9
          |vpiLhs:
          \_ref_obj: (work@top.named_block.foo), line:4:5, endln:4:8, parent:work@top.named_block
            |vpiName:foo
            |vpiFullName:work@top.named_block.foo
            |vpiActual:
            \_logic_var: (work@top.named_block.foo), line:3:14, endln:3:17, parent:work@top.named_block
    |vpiAlwaysType:1
|uhdmtopModules:
\_module: work@top (work@top) width_unknown.v:1:1: , endln:6:10
  |vpiName:work@top
  |vpiDefName:work@top
  |vpiTop:1
  |vpiTopModule:1
```

Unclear if this needed to be reflected with fully qualified name in UHDM (in which case CC: @alaindargelas ) or if this needs to be translated differently in the plugin. Put here to triage properly.

alaindargelas, 2022-05-11:
A recent change I did duplicated this var:  \_logic_var: (work@top.named_block.foo)
as you can see there are 2 entries. I fixed that with [#2926](https://github.com/chipsalliance/Surelog/pull/2926).
That could be the problem. The UHDM dump looks ok besides that.

hzeller, 2022-05-11:
This looks like to be working now, thanks!

closed 2022-05-11 by hzeller