chipsalliance/yosys-f4pga-plugins #326: Build UHDM plugin error

minghungumich, 2022-05-19:
I was trying this repo https://github.com/antmicro/yosys-uhdm-plugin-integration to add system verily plugin into yosys.
But when I try to run the build_binaries.sh script, it stuck at UHDM plugin part. I believe it is related to this repo. https://github.com/chipsalliance/yosys-f4pga-plugins
The error message is listed below. The compiler is complaining some file missing, but I guess it might be g++ version mismatch. We are currently using GCC 12.1.0.
Since the plugin is not installed correctly, when doing “plugin -i systemverilog”, it will display “ERROR: Can’t load module”. 
Could you show your environment for the correct build? Or there is another prebuilt binary could be directly used? 

[sdc.cc](http://sdc.cc/): In member function ‘std::vector<std::__cxx11::basic_string<char> > {anonymous}::GetClocksCmd::extract_list(const std::string&)’:
[sdc.cc:224](http://sdc.cc:224/):14: error: ‘istream_iterator’ is not a member of ‘std’
  224 |         std::istream_iterator<std::string> begin(ss);
      |              ^~~~~~~~~~~~~~~~
[sdc.cc:30](http://sdc.cc:30/):1: note: ‘std::istream_iterator’ is defined in header ‘<iterator>’; did you forget to ‘#include <iterator>’?
   29 | #include "set_max_delay.h"
  +++ |+#include <iterator>
   30 |
[sdc.cc:224](http://sdc.cc:224/):42: error: expected primary-expression before ‘>’ token
  224 |         std::istream_iterator<std::string> begin(ss);
      |                                          ^
[sdc.cc:224](http://sdc.cc:224/):49: error: no matching function for call to ‘begin(std::stringstream&)’
  224 |         std::istream_iterator<std::string> begin(ss);
      |                                            ~~~~~^~~~
In file included from /usr/local/include/c++/12.1.0/bits/algorithmfwd.h:39,
                 from /usr/local/include/c++/12.1.0/bits/stl_algo.h:59,
                 from /usr/local/include/c++/12.1.0/algorithm:61,
                 from [sdc.cc:18](http://sdc.cc:18/):
/usr/local/include/c++/12.1.0/initializer_list:90:5: note: candidate: ‘template<class _Tp> constexpr const _Tp* std::begin(initializer_list<_Tp>)’
   90 |     begin(initializer_list<_Tp> __ils) noexcept
      |     ^~~~~
/usr/local/include/c++/12.1.0/initializer_list:90:5: note:   template argument deduction/substitution failed:
[sdc.cc:224](http://sdc.cc:224/):49: note:   ‘std::__cxx11::basic_stringstream<char>’ is not derived from ‘std::initializer_list<_Tp>’
  224 |         std::istream_iterator<std::string> begin(ss);
      |                                            ~~~~~^~~~
In file included from /usr/local/include/c++/12.1.0/array:44,
                 from [sdc.cc:19](http://sdc.cc:19/):
/usr/local/include/c++/12.1.0/bits/range_access.h:52:5: note: candidate: ‘template<class _Container> decltype (__cont.begin()) std::begin(_Container&)’
   52 |     begin(_Container& __cont) -> decltype(__cont.begin())
      |     ^~~~~
/usr/local/include/c++/12.1.0/bits/range_access.h:52:5: note:   template argument deduction/substitution failed:
/usr/local/include/c++/12.1.0/bits/range_access.h: In substitution of ‘template<class _Container> decltype (__cont.begin()) std::begin(_Container&) [with _Container = std::__cxx11::basic_stringstream<char>]’:
[sdc.cc:224](http://sdc.cc:224/):49:   required from here
/usr/local/include/c++/12.1.0/bits/range_access.h:52:50: error: ‘class std::__cxx11::basic_stringstream<char>’ has no member named ‘begin’; did you mean ‘beg’?
   52 |     begin(_Container& __cont) -> decltype(__cont.begin())
      |                                           ~~~~~~~^~~~~
      |                                           beg
/usr/local/include/c++/12.1.0/bits/range_access.h:63:5: note: candidate: ‘template<class _Container> decltype (__cont.begin()) std::begin(const _Container&)’
   63 |     begin(const _Container& __cont) -> decltype(__cont.begin())
      |     ^~~~~
/usr/local/include/c++/12.1.0/bits/range_access.h:63:5: note:   template argument deduction/substitution failed:
/usr/local/include/c++/12.1.0/bits/range_access.h: In substitution of ‘template<class _Container> decltype (__cont.begin()) std::begin(const _Container&) [with _Container = std::__cxx11::basic_stringstream<char>]’:
[sdc.cc:224](http://sdc.cc:224/):49:   required from here
/usr/local/include/c++/12.1.0/bits/range_access.h:63:56: error: ‘const class std::__cxx11::basic_stringstream<char>’ has no member named ‘begin’; did you mean ‘beg’?
   63 |     begin(const _Container& __cont) -> decltype(__cont.begin())
      |                                                 ~~~~~~~^~~~~
      |                                                 beg
/usr/local/include/c++/12.1.0/bits/range_access.h:95:5: note: candidate: ‘template<class _Tp, long unsigned int _Nm> _Tp* std::begin(_Tp (&)[_Nm])’
   95 |     begin(_Tp (&__arr)[_Nm]) noexcept
      |     ^~~~~
/usr/local/include/c++/12.1.0/bits/range_access.h:95:5: note:   template argument deduction/substitution failed:
[sdc.cc:224](http://sdc.cc:224/):49: note:   mismatched types ‘_Tp [_Nm]’ and ‘std::stringstream’ {aka ‘std::__cxx11::basic_stringstream<char>’}
  224 |         std::istream_iterator<std::string> begin(ss);
      |                                            ~~~~~^~~~
[sdc.cc:225](http://sdc.cc:225/):14: error: ‘istream_iterator’ is not a member of ‘std’
  225 |         std::istream_iterator<std::string> end;
      |              ^~~~~~~~~~~~~~~~
[sdc.cc:225](http://sdc.cc:225/):14: note: ‘std::istream_iterator’ is defined in header ‘<iterator>’; did you forget to ‘#include <iterator>’?
[sdc.cc:225](http://sdc.cc:225/):42: error: expected primary-expression before ‘>’ token
  225 |         std::istream_iterator<std::string> end;
      |                                          ^
[sdc.cc:225](http://sdc.cc:225/):44: error: ‘end’ was not declared in this scope; did you mean ‘std::end’?
  225 |         std::istream_iterator<std::string> end;
      |                                            ^~~
      |                                            std::end
/usr/local/include/c++/12.1.0/bits/range_access.h:106:5: note: ‘std::end’ declared here
  106 |     end(_Tp (&__arr)[_Nm]) noexcept
      |     ^~~
[sdc.cc:226](http://sdc.cc:226/):19: error: ‘begin’ was not declared in this scope; did you mean ‘std::begin’?
  226 |         std::copy(begin, end, std::back_inserter(port_list));
      |                   ^~~~~
      |                   std::begin
/usr/local/include/c++/12.1.0/bits/range_access.h:95:5: note: ‘std::begin’ declared here
   95 |     begin(_Tp (&__arr)[_Nm]) noexcept
      |     ^~~~~
[sdc.cc](http://sdc.cc/): In constructor ‘{anonymous}::SdcPlugin::SdcPlugin()’:
[sdc.cc:337](http://sdc.cc:337/):34: warning: member ‘{anonymous}::SdcPlugin::sdc_writer_’ is used uninitialized [-Wuninitialized]
  337 |     SdcPlugin() : write_sdc_cmd_(sdc_writer_), set_false_path_cmd_(sdc_writer_), set_max_delay_cmd_(sdc_writer_), set_clock_groups_cmd_(sdc_writer_)
      |                                  ^~~~~~~~~~~
[sdc.cc:337](http://sdc.cc:337/):68: warning: member ‘{anonymous}::SdcPlugin::sdc_writer_’ is used uninitialized [-Wuninitialized]
  337 |     SdcPlugin() : write_sdc_cmd_(sdc_writer_), set_false_path_cmd_(sdc_writer_), set_max_delay_cmd_(sdc_writer_), set_clock_groups_cmd_(sdc_writer_)
      |                                                                    ^~~~~~~~~~~
[sdc.cc:337](http://sdc.cc:337/):101: warning: member ‘{anonymous}::SdcPlugin::sdc_writer_’ is used uninitialized [-Wuninitialized]
  337 |     SdcPlugin() : write_sdc_cmd_(sdc_writer_), set_false_path_cmd_(sdc_writer_), set_max_delay_cmd_(sdc_writer_), set_clock_groups_cmd_(sdc_writer_)
      |                                                                                                     ^~~~~~~~~~~
[sdc.cc:337](http://sdc.cc:337/):137: warning: member ‘{anonymous}::SdcPlugin::sdc_writer_’ is used uninitialized [-Wuninitialized]
  337 |     SdcPlugin() : write_sdc_cmd_(sdc_writer_), set_false_path_cmd_(sdc_writer_), set_max_delay_cmd_(sdc_writer_), set_clock_groups_cmd_(sdc_writer_)
      |                                                                                                                                         ^~~~~~~~~~~
make[1]: *** [sdc.o] Error 1
make[1]: Leaving directory `/home/kevinchen/yosys-f4pga-plugins/sdc-plugin'
make: *** [install_sdc] Error 2

minghungumich, 2022-05-19:
@mithro @msaligine

tgorochowik, 2022-05-20:
Thanks for the report!

It looks as thought the SDC plugin is failing to build. We will take a look at this, there are two actions I think are needed:
* See if the SDC plugin can be built with gcc-12 - fix it with not
* make sure we don't unnecessarily build all plugins in the uhdm integration repository, as they are not installed there anyway

In the meantime, to answer your other question - yes, to make it easier we provide binary builds on the releases page of the integration repo: https://github.com/antmicro/yosys-uhdm-plugin-integration/releases

Please feel free to download the latest yosys-uhdm-plugin-xxx package - it contains the plugin and an installation script - you need yosys installed separately if you go this way (either built and installed from sources or from a package if you prefer).

Please let us know if this is enough for you to get started.

tgorochowik, 2022-05-20:
> What is the tool environment for the correct build?

You can see the dependencies being set up here:
https://github.com/antmicro/yosys-uhdm-plugin-integration/blob/2cd481dab4c40696a5b14763eae1ef302619a1d6/.github/workflows/main.yml#L22-L31

And the build is started here:
https://github.com/antmicro/yosys-uhdm-plugin-integration/blob/2cd481dab4c40696a5b14763eae1ef302619a1d6/.github/workflows/main.yml#L51-L63

So gcc-9 is used currently, it should work for you too, but we'll have to address the gcc-12 issue anyway.

tgorochowik, 2022-05-20:
The error in the SDC plugin seems to be this addressed here: #324

minghungumich, 2022-05-20:
Thanks @tgorochowik ! I am able to install the system verilog plugin in yosys from the release packages.

minghungumich, 2022-05-22:
@tgorochowik @mithro @msaligane Currently I can use system verily plugin with example on my host. But when I want to install this in my docker, I found it needs higher version of libstdc++so.6 (`GLIBCXX_3.4.20' not found, I only have up to 19). And it is not supported in my original docker image, so I want to ask do we have the docker image that install all of this tool inside? I am currently using enables/openlane https://hub.docker.com/r/efabless/openlane

umarcor, 2022-05-23:
@minghungumich there are containers maintained in [hdl/containers](https://github.com/hdl/containers) which provide a preinstalled environment to work with F4PGA tools. See [hdl.github.io/containers: Tools and images » F4PGA (Conda)](https://hdl.github.io/containers/ToolsAndImages.html#f4pga-conda). Those are created using the Conda environments from [f4pga-examples](https://github.com/chipsalliance/f4pga-examples) (https://github.com/chipsalliance/f4pga-examples/blob/main/xc7/environment.yml or https://github.com/chipsalliance/f4pga-examples/blob/main/eos-s3/environment.yml), so the yosys-plugins are included: https://github.com/chipsalliance/f4pga-examples/blob/main/xc7/environment.yml#L11. In https://github.com/hdl/packages/blob/main/.github/workflows/containers-conda-f4pga.yml you will find examples of how to use/run those containers (see executions in CI: https://github.com/hdl/packages/actions/workflows/containers-conda-f4pga.yml).

minghungumich, 2022-05-25:
@umarcor @tgorochowik @mithro @msaligane I am currently trying to use OpenROAD-flow-scripts with system verilog plugin. https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/synth_preamble.tcl#L23 I modify this line from read_verilog $file to read_systemverilog $file. Besides, I insert "plugin -i systemverilog" before reading system verilog file. The log displays ERROR: TCL interpreter returned an error: invalid command name "read_systemverilog", but I can actually pass this when I log into the yosys, import plugin, and read file. I am curious how to modify the ORFS script to support reading system verilog function.

msaligane, 2022-05-25:
@minghungumich can you first see what are the issues with the synthesized netlist first before modifying any of the flows? 
We can always tweak ORFS or OL to read in SV-based designs when the above is done.

rkapuscik, 2022-05-26:
As a note, for now the `read_systemverilog` command requires all files to be parsed at the same time, the script calls `read_verilog` separately for each file so simple substitution won't work here.
Support for separate compilation is being worked on, only recently Surelog made it possible as described in #331.