chipsalliance/yosys-f4pga-plugins #339: How to build tcl script file for yosys systemverilog plugin?

minghungumich, 2022-05-27:
As #326 points, I can reproduce the systemverilog result in #295 in yosys interactive mode.
Like the script below:
`plugin -i systemverilog
read_systemverilog $file
synth -top $top_module
write_verilog $netlist`

I can also put those command as a script and run `yosys -s script_file`

But when I try to do the same thing in the tcl script file, it cannot identify the read_systemverilog command.
Here is my tcl script.
`yosys -import
plugin -i systemverilog
read_systemverilog $file
synth -top $top_module
write_verilog $netlist`

minghungumich, 2022-05-27:
@rkapuscik @kgugala @hzeller @umarcor @tgorochowik @mithro @msaligane Now I can build the systemverilog plugin but I face issue with running those commands as a tcl script file.

msaligane, 2022-05-29:
@minghungumich can you please add more description to this issue

minghungumich, 2022-05-29:
@msaligane The issue has been updated.

mithro, 2022-05-29:
@kgugala / @hzeller - Can you help figure out what is going on here and how the documentation needs to be updated so other people don't run into the same issue?

rkapuscik, 2022-05-30:
This is caused by a somewhat counterintuitive way Yosys commands are added to tcl - and it affects every plugin. You need to re-import the commands after adding the plugin to gain access to them in tcl, like this:
```tcl
# Import base Yosys commands
yosys -import
# Load plugin(s)
plugin -i systemverilog
# Import commands added by the plugin
yosys -import
# Further steps...
read_systemverilog $file
```
You can also take a look at how it's done in the [tests](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/main/systemverilog-plugin/tests/counter/counter.tcl) for the [plugins](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/main/sdc-plugin/tests/pll/pll.tcl#L1-L3).

minghungumich, 2022-06-02:
Thanks! That change works!

closed 2022-06-02 by minghungumich