chipsalliance/yosys-f4pga-plugins #341: Issue related to the BRAM inference

lilitrs, 2022-05-31:
After latest changes in synth_ql regarding the BRAM inference got the following issue for qlf_k6n10f family:

BRAM: RegFilePlugin_regFile.0.0.0 ($__RSF_FACTOR_BRAM18_TDP) + RegFilePlugin_regFile.1.0.0 ($__RSF_FACTOR_BRAM18_TDP) => bram_RegFilePlugin_regFile.0.0.0_RegFilePlugin_regFile.1.0.0 (BRAM2x18_TDP)
terminate called after throwing an instance of 'std::out_of_range'
  what():  Cell::getParam()

The test benchmarks are attached.
[ethernet_mac.zip](https://github.com/chipsalliance/yosys-f4pga-plugins/files/8803533/ethernet_mac.zip) - top_module = eth_top
[VexRiscv.zip](https://github.com/chipsalliance/yosys-f4pga-plugins/files/8803535/VexRiscv.zip) - top_module = VexRiscv

mkurc-ant, 2022-06-06:
Thank you for reporting the issue. I've managed to reproduce it locally. Working on a solution.

lpawelcz, 2022-06-14:
Hi @lilitrs, I've opened a PR: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/348 that fixes your problem. I tested it on both of your designs and it looks OK now.

lilitrs, 2022-07-18:
Verified. Closing the issue as it is fixed.

closed 2022-07-18 by lilitrs