chipsalliance/yosys-f4pga-plugins #345: Systemverilog-plugin fails to identify macro function

minghungumich, 2022-06-02:
In my design, there is a definition for macro function as shown below.
```
`define log2(VALUE) ((VALUE) < ( 1 ) ? 0 : (VALUE) < ( 2 ) ? 1 : (VALUE) < ( 4 ) ? 2 : (VALUE) < ( 8 ) ? 3 : (VALUE) < ( 16 )  ? 4 : (VALUE) < ( 32 )  ? 5 : (VALUE) < ( 64 )  ? 6 : (VALUE) < ( 128 ) ? 7 : (VALUE) < ( 256 ) ? 8 : (VALUE) < ( 512 ) ? 9 : (VALUE) < ( 1024 ) ? 10 : (VALUE) < ( 2048 ) ? 11 : (VALUE) < ( 4096 ) ? 12 : (VALUE) < ( 8192 ) ? 13 : (VALUE) < ( 16384 ) ? 14 : (VALUE) < ( 32768 ) ? 15 : (VALUE) < ( 65536 ) ? 16 : (VALUE) < ( 131072 ) ? 17 : (VALUE) < ( 262144 ) ? 18 : (VALUE) < ( 524288 ) ? 19 : (VALUE) < ( 1048576 ) ? 20 : (VALUE) < ( 1048576 * 2 ) ? 21 : (VALUE) < ( 1048576 * 4 ) ? 22 : (VALUE) < ( 1048576 * 8 ) ? 23 : (VALUE) < ( 1048576 * 16 ) ? 24 : 25)
```
I check this can be compiled in design compiler successfully, but fails in this plugin as shown below. Do I have another alternative for this macro function or there is other option command needs to be added.
```
[ERR:CP0337] ./designs/src/pulpino_yosys/ips/apb/apb_spi_master/apb_spi_master.sv:24:35: Undefined user task/function "log2".

[ERR:UH0724] ./designs/src/pulpino_yosys/ips/apb/apb_spi_master/apb_spi_master.sv:24:35: Undefined user task/function "log2".
```

minghungumich, 2022-06-02:
@mithro @msaligane I face another issue that yosys cannot identify for this type of macro function.

rkapuscik, 2022-06-03:
Thanks for reporting this.

Generally errors with format like that (`[ERR:XXXX] ...`) come from Surelog, but I couldn't reproduce the one you reported. Can you share the file where you use this macro (or a minimal example) and a script fragment where you read the definition and the file?

Also, maybe using `$clog2()` would work for you? It should be supported by both Surelog and the plugin.

minghungumich, 2022-06-03:
> Thanks for reporting this.
>
> Generally errors with format like that (`[ERR:XXXX] ...`) come from Surelog, but I couldn't reproduce the one you reported. Can you share the file where you use this [...]

@rkapuscik I tried `$clog2()` works for my case. This is [system verilog file](https://github.com/minghungumich/OpenROAD-flow-scripts/blob/master/flow/designs/src/pulpino_yosys/ips/apb/apb_spi_master/apb_spi_master.sv) defining the macro and [script](https://github.com/minghungumich/OpenROAD-flow-scripts/blob/master/flow/scripts/synth_systemverilog.tcl) I run for yosys plugin.

minghungumich, 2022-06-03:
@rkapuscik I just found another issue after fixing `log2` macro error. [This file](https://github.com/minghungumich/OpenROAD-flow-scripts/blob/master/flow/designs/src/pulpino_yosys/ips/apb/apb_event_unit/generic_service_unit.sv#L59) uses break statement in the for loop. It should be synthesizable but it displays error message below.
```
ERROR: ./designs/src/pulpino_yosys/ips/apb/apb_event_unit/generic_service_unit.sv:59: Encountered unhandled object '' of type 'break_stmt'
```