chipsalliance/yosys-f4pga-plugins #366: Yosys-plugin: use-after-delete if unused module removed

hzeller, 2022-07-15:
Running the plugin with [address sanitizer](https://clang.llvm.org/docs/AddressSanitizer.html) revealed that there are situations that deleted objects are re-used.

I hot-fixed it locally by skipping the deleted values, but I suspect a proper fix would possibly removing it from the container; leaving it up to you.

```patch
--- a/systemverilog-plugin/UhdmAst.cc
+++ b/systemverilog-plugin/UhdmAst.cc
@@ -598,6 +598,7 @@ static AST::AstNode *convert_dot(AST::As
 static void setup_current_scope(std::unordered_map<std::string, AST::AstNode *> top_nodes, AST::AstNode *current_top_node)
 {
     for (auto it = top_nodes.begin(); it != top_nodes.end(); it++) {
+        if (!it->second) continue;
         if (it->second->type == AST::AST_PACKAGE) {
             for (auto &o : it->second->children) {
                 // import only parameters
@@ -1408,7 +1409,7 @@ void UhdmAst::process_design()
         }
     }
     // Once we walked everything, unroll that as children of this node
-    for (auto pair : shared.top_nodes) {
+    for (auto &pair : shared.top_nodes) {
         if (!pair.second)
             continue;
         if (!pair.second->get_bool_attribute(UhdmAst::partial())) {
@@ -1424,6 +1425,7 @@ void UhdmAst::process_design()
         } else {
             log_warning("Removing unused module: %s from the design.\n", pair.second->str.c_str());
             delete pair.second;
+            pair.second = nullptr;
         }
     }
 }
```

We really need https://github.com/antmicro/yosys-uhdm-plugin-integration/issues/649 to find these things early.

hzeller, 2022-07-26:
Any update on this one ?

rkapuscik, 2022-07-26:
This is being worked on as part of https://github.com/antmicro/yosys-uhdm-plugin-integration/pull/913 - I'll try to split that PR to get some of the changes merged faster.