chipsalliance/yosys-f4pga-plugins #372: Document how to run the plugins, and how to contribute.

kanndil, 2022-07-21:
I am working on a plugin, and am trying to figure out how to add it to this repo. 

I can't figure out whether this repo is just a collection of plugins, and each plugin works independently and is built independently,
or it implements a way of building all the plugins at once. And if so, what steps should I follow before I add my plugin, and pull request.

tmichalak, 2022-07-22:
@kanndil glad to hear that you are working on a plugin that clearly planning to contribute by opening a PR.
This repo holds the sources and tests for the plugins so you came to the right place :) 
We don't have a step-by-step guide in the documentation, but you can use any of the existing directories/plugins as reference (such as the [sdc-plugin](https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/sdc-plugin). Each of those plugins contains a main Makefile that includes the [Makefile_plugin.common](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/main/Makefile_plugin.common) that has a nice description of the structure of the plugin. From there you will see that another characteristic of the plugin directory is the tests directory that has a separate Makefile which includes a [Makefile_test.common](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/main/Makefile_test.common). 
So, basically a new plugin should go into a separate directory and consist of the plugins's source code, tests and a handful of makefiles which include the `Makefile_plugin.common` and `Makefile_test.common`.

kanndil, 2022-07-23:
@tmichalak  Thank you for you response. 

I followed the instructions, however still having a problem running the makefile and the (plugin -i) command.  

When I run the makefile, I expect a plugin_name.so file to be added to Yosys's shared folder, but the file remains empty.

kanndil, 2022-07-26:
Apparently I have the same problem in issue #74

tmichalak, 2022-07-26:
@kanndil, but you are able to build the .so file, correct? Do you call `make install` ?

kanndil, 2022-07-26:
@tmichalak yes, the building works fine, only the copying to the shared folder part doesn't work

tmichalak, 2022-07-27:
@kanndil have you tried `install -d` ?

kanndil, 2022-07-27:
@tmichalak removing the -D option worked for me, the shared object file is copied successfully to the directory
[image]