chipsalliance/yosys-f4pga-plugins #380: verilog_defaults doesn't work with systemverilog plugin

dpetrisko, 2022-08-25:
Hi, not sure if this the right place to ask.

I'm trying to use the systemverilog plugin for yosys, but I'm having trouble with include paths. The UHDM frontend doesn't seem to pick up the verilog_defaults.

yosys> plugin -i systemverilog
yosys> verilog_defaults -add -Ibasejump_stl/bsg_misc
yosys> read_systemverilog basejump_stl/bsg_cache/bsg_cache_pkg.v

```
yosys> read_systemverilog basejump_stl/bsg_cache/bsg_cache_pkg.v

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[ERR:PP0101] basejump_stl/bsg_cache/bsg_cache_pkg.v:7: Cannot open include file "bsg_defines.v".

[WRN:PA0205] basejump_stl/bsg_cache/bsg_cache_pkg.v:9:1: No timescale set for "bsg_cache_pkg".

[INF:CP0300] Compilation...

[INF:CP0301] basejump_stl/bsg_cache/bsg_cache_pkg.v:9:1: Compile package "bsg_cache_pkg".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[NTE:EL0508] Nb Top level modules: 0.

[NTE:EL0509] Max instance depth: 0.

[NTE:EL0510] Nb instances: 0.

[NTE:EL0511] Nb leaf instances: 0.

[INF:UH0706] Creating UHDM Model...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 1
[WARNING] : 1
[   NOTE] : 4
ERROR: Error when parsing design. Aborting!
```

On the other hand, this works: 
```
yosys> read_systemverilog -Ibasejump_stl/bsg_misc basejump_stl/bsg_cache/bsg_cache_pkg.v
1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:59:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:73:9: Unused macro argument "val".

[WRN:PP0113] basejump_stl/bsg_misc/bsg_defines.v:124:9: Unused macro argument "x".

[WRN:PA0205] basejump_stl/bsg_cache/bsg_cache_pkg.v:9:1: No timescale set for "bsg_cache_pkg".

[INF:CP0300] Compilation...

[INF:CP0301] basejump_stl/bsg_cache/bsg_cache_pkg.v:9:1: Compile package "bsg_cache_pkg".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[NTE:EL0508] Nb Top level modules: 0.

[NTE:EL0509] Max instance depth: 0.

[NTE:EL0510] Nb instances: 0.

[NTE:EL0511] Nb leaf instances: 0.

[INF:UH0706] Creating UHDM Model...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 4
[   NOTE] : 4
```

But my project has dozens of include directories, so this doesn't seem like a scalable solution. Is it possible to parse verilog_defaults for includes and use them in the UHDM frontend?

tgorochowik, 2022-08-27:
This should be fixed by
#377 

It adds SV equipments of those commands

mglb, 2023-05-15:
Seems to be fixed, closing.

closed 2023-05-15 by mglb