chipsalliance/yosys-f4pga-plugins #400: systemverilog-plugin: `enum`s are always 32-bit

mglb, 2022-11-16:
Enums like `enum logic [2:0]` are seen as 32 bit wide types. This is a problem only inside the plugin - yosys converts them to properly sized types in later stages.

Test case:
```systemverilog
module mod(output int foo_size, output int bar_size, output int enum_size);
  typedef enum logic [2:0] {
      e_zero  = 0
    , e_one   = 1
    , e_two   = 2
    , e_three = 3
    , e_four  = 4
  } numbers_e;

  typedef struct packed {
    logic [31:0] f_member;
  } foo_t;

  typedef struct packed {
    numbers_e    b_member;
    logic [28:0] b_padding;
  }  bar_t;

  union packed {
    foo_t member_foo;
    bar_t member_bar;
  } union_s;

  assign foo_size = $bits(foo_t);
  assign bar_size = $bits(bar_t);
  assign enum_size = $bits(numbers_e);
endmodule
```
causes an error:
```
enum_width.sv:10: ERROR: member member_foo of a packed union has 32 bits, expecting 61
```

After commenting out the union and using `write_verilog` in yosys the sizes are correct:
```
  assign foo_size = 32'h00000020;
  assign bar_size = 32'h00000020;
  assign enum_size = 32'h00000003;
```

taylor-bsg, 2022-12-03:
@mglb any insights on this one?

mithro, 2022-12-03:
@hzeller - Can you chat with Antmicro about this?

taylor-bsg, 2022-12-06:
> @taylor-bsg <https://github.com/taylor-bsg> the PR with a fix will be
> ready today: #416

Awesome!

closed 2022-12-12 by mglb