chipsalliance/yosys-f4pga-plugins #412: Yosys verilog reader enhancement request

shakouri, 2022-11-30:
The yosys verilog reader is not able to parse verilog code that other synthesis tools are able to parse. sv2v plugin is not working either.

1.  Allow parameter passing w/o use of parantheses ('-' denotes original RTL, and '+' denotes changes to make yosys read it in):
[image]

2. Allow use of ? during async portion of always block ('-' denotes original RTL, and '+' denotes changes to make yosys read it in):
[image]
[image]

3. Allow use of logic as instance name('-' denotes original RTL, and '+' denotes changes to make yosys read it in):
[image]

mithro, 2022-11-30:
@shakouri - I was actually talking about;

> I am getting the following error when I try to use system verilog plugin:
> ```
> yosys> plugin -i systemverilog
> ERROR: Can't load module `./systemverilog': dlopen(/usr/local/bin/../share/yosys/plugins/systemverilog.so, 5): no suitable image found. Did find:
> /usr/local/bin/../share/yosys/plugins/systemverilog.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x03
> /usr/local/Cellar/yosys/0.23/share/yosys/plugins/systemverilog.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x03
> ```

shakouri, 2022-11-30:
Thanks. If someone can tell me how to fix this error so I can try this plugin also that would be great!.

pgielda, 2022-11-30:
@mithro 
"0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x03" means:
0x7F 0x45 0x4C 0x46 == 0x7F "ELF" [ELF HEADER]
0x02 => 64-bit binary
0x01 => Little Endian
0x03 => Linux

Maybe the host is not a 64-bit Little Endian Linux?

shakouri, 2022-11-30:
I am running on a mac m1 processor. Where should I get the plugin for it?

pgielda, 2022-11-30:
You're running Linux on Mac M1, or MacOS / OSX?

pgielda, 2022-11-30:
(In any case, if the plugin was not compiled from scratch, it will not work on ARM. If its MacOS -- ELF is not even the format that MacOS uses, MacOS uses dylib files).

shakouri, 2022-11-30:
It is a MacOS using M1 processor. I have not been able to compile it from scratch, so I was trying to download it if the plugin exists for MacOS.

pgielda, 2022-11-30:
Not yet unfortunately, current focus is Linux. We provide binaries for x86_64 and I am pretty sure you would be able to compile from scratch on ARM64 + Linux but we're currently not targetting MacOS.

We do plan to change that long term but short term no other solution than to try to compile from sources and then fix any issues that occur.

shakouri, 2022-11-30:
Is it possible to check if the three verilog cases work with UHDM plugin?

rkapuscik, 2022-12-06:
I ran the cases through the plugin with following results:
1. Is parsed, but the parameter value is ignored (not present in UHDM). Looking at the LRM (section A.4.1.1 Module Instantiation) I don't think omitting parentheses is legal in this context, but if you have an example of tool supporting this I suggest opening an issue in the [Surelog repository](https://github.com/chipsalliance/Surelog/) (which is the tool we use for parsing), as it would need to be added there as a first step.
2. Is accepted both by Yosys and the plugin, but fails during processing in Yosys with `ERROR: Multiple edge sensitive events found for this signal!`. Unfortunately this doesn't seem to be fixable inside the plugin and would need to be resolved in Yosys itself.
3. Fails during parsing in Surelog with 
```
[SNT:PA0207] ./syntax.sv:15:8: Syntax error: no viable alternative at input 'sparc_exu_alulogic logic',
```
Similar to 1., if you have an example of a tool that supports this please open an issue in the Surelog repository.

shakouri, 2022-12-09:
I have opened a case in the Surelog repository: https://github.com/chipsalliance/Surelog/issues/3371