chipsalliance/yosys-f4pga-plugins #414: systemverilog plugin : assert error

shushruthholla, 2022-12-05:
I have been working on symbiyosys for formal tests along with it I am using the systemverilog plugin. Recently while runnig the sby script for a design, I encountered an assert error after elaboration as seen below
``ERROR: Assert `AST_INTERNAL::current_scope.count(wiretype_node->str)' failed UhdmAst.cc:304.``
It would be helpful to know what is causing it and if there is any fix.

mithro, 2022-12-06:
@kgugala - Who should look at this?

shushruthholla, 2022-12-07:
Is there any update?

shushruthholla, 2022-12-09:
Is there anyone looking into this error? If any additional thing is required from my side I can post it. I would be really helpful if I get a reply from someone.

rkapuscik, 2022-12-09:
This error means that the plugin expected a node to refer to a named type, but the name was missing - most probably because the plugin created a wrong type of node in a previous step. Can you share the design where you observe this error?

shushruthholla, 2022-12-09:
Thanks for the response. I will share the link to the design. The main design files I have used are pmp_top.sv and pmp96_yosys.v. The pmp96_yosys.v is a yosys synthesized file of pmp96.vhdl. The sby file is pmp is pmp.sby. I have used the open-source version of symbiyosys for verification. (I am aware that the open source version does not support concurrent assertions)

Link: 
https://github.com/shushruthholla/pmp_formal-verification.git

shushruthholla, 2022-12-12:
Was I able to make myself clear in the previous comment?

rkapuscik, 2022-12-13:
Yes, thank you. I've reproduced the failure. It's related to incorrect processing of typedefs outside module scope, we'll look into this.
As a temporary workaround, I verified that `pmp_top.sv` can be parsed after moving the typedefs inside module.

You also mentioned that you were interested in formal verification. Plugin only partially supports this for now, feel free to open more issues if any constructs you need are missing.

shushruthholla, 2022-12-15:
Thanks for the response. Also, I will be opening more issues if I find new ones. Lastly, do you mind linking the changes that you did for 'pmp_top.sv'?

rkapuscik, 2022-12-22:
Thanks for your patience! #427 should resolve the original issue. The changes I made were literally just moving the typedefs, but that shouldn't be needed after that PR is merged.

I'm not sure about the expected symbiyosys behavior, but the plugin should error out after encountering a syntax error or unsupported statement. We are however relying on Surelog to include the correct constructs in UHDM, so maybe it's skipped there. Can you point to an example of the code that's skipped?

shushruthholla, 2022-12-22:
Thanks for the response. I was using the ´pmp_top.sv´ as the top file for symbiyosys verification with the script file being ´pmp.sby´. The ´pmp_top.sv´  file has assert and assume concurrent property assertions. I tried disabling the assume properties to see if the tool would report any error. Also I tried  commeting out ´prv == `M_MODE &&´ (line 330) in the ´property m_always_grant´ (line 325) since this property requires M mode to operate. I think both of these situations should give an error but the tool still gives a pass for the verification check.

shushruthholla, 2023-01-03:
can give a review about my previous comment @rkapuscik ?

kgugala, 2023-01-04:
it looks like the issue here was closed automatically when we merged the PR. I'm reopenign this as @shushruthholla has some questions

tgorochowik, 2023-03-15:
This should be fixed by #463 and #454