chipsalliance/yosys-f4pga-plugins #467: Currently support for custom-type with range is limited to single range.

grotival, 2023-03-09:
`Currently support for custom-type with range is limited to single range` message is emitted from [this line](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/24755e3b43400ad25e90fe01cc764a28a12de999/systemverilog-plugin/UhdmAst.cc#L821) in `yosys-f4pga-plugins` in this minimal repro case:

```
package repro_pkg;

  parameter int J = 3;
  parameter int K = 2;

  typedef struct packed {
    logic a;
    logic b;
    logic c;
  } my_packed_struct_t;

  typedef union packed {
    my_packed_struct_t [K:0][J:0] my_array;
    my_packed_struct_t [K:0][J:0] my_array2;
  } my_packed_struct_with_array_t;

endpackage : repro_pkg

module repro;
endmodule : repro
```

Maybe this should be added to [the testbench](https://chipsalliance.github.io/sv-tests-results/)
As far as I can tell [this case](https://github.com/chipsalliance/sv-tests/blob/3589dc3804673208c6edbd9577e856516b087f78/tests/chapter-7/arrays/multidimensional/basic.sv) is the closest but it's only for a user-defined type containing an array. The problem is when using an array of a user-defined type.

grotival, 2023-03-09:
We appreciated if this could be implemented as this is quite a common case at Google. Chatted with @QuantamHD about this issue who told me maybe you could get this prioritised?

thanks a lot!

QuantamHD, 2023-03-17:
@kamilrakoczy is this something you could look at?

kamilrakoczy, 2023-03-20:
Sure, I'll look at this.

kamilrakoczy, 2023-04-05:
@grotival After https://github.com/chipsalliance/yosys-f4pga-plugins/pull/470 plugin should support multi-range user-defined type inside struct/union. Could you please test if this change works for you?

closed 2023-04-05 by kamilrakoczy

grotival, 2023-04-12:
Will do!
thanks!

grotival, 2023-05-02:
verified on my side, all good now, thanks a lot!