chipsalliance/yosys-f4pga-plugins #496: synth_quicklogic name conflict

hzeller, 2023-04-26:
There is a `synth_quicklogic` pass in [yosys: techlibs/quicklogic](https://github.com/YosysHQ/yosys/blob/master/techlibs/quicklogic/synth_quicklogic.cc) as well  as in [yosys-f4pga-plugins: ql-qlf](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/main/ql-qlf-plugin/synth_quicklogic.cc).

If Yosys has its version compiled in, it is not possible to load the yosys-f4pga-plugin, as a pass with that name is already registered:
```
ERROR: Unable to register pass 'synth_quicklogic', pass already exists!
```

Not sure how to resolve, probably our pass needs to be renamed to not clash with the other name ?

hzeller, 2023-04-26:
Ah, looks like @timkpaine is working on it in https://github.com/chipsalliance/yosys-f4pga-plugins/pull/485/files#diff-85e2fb37a6725033c760e3da33ad76a89c4da53b332591cb70e403a2a4744d68

timkpaine, 2023-04-26:
Rename did the trick

hzeller, 2023-04-28:
Assigning to Karol to triage.