chipsalliance/yosys-f4pga-plugins #502: sv_plugin: unhandled typespec in process_typespec_member:

QuantamHD, 2023-05-02:
When running the following example we get the error `hardware/tools/open_road/surelog_testcases/struct.sv:27 Encountered unhandled typespec in process_typespec_member: 'foo_flags::common_flags_t' of type 'unsupported_typespec'`

```systemverilog
package foo_flags;
  typedef struct packed {
    logic a;
    logic b;
    logic c;
  } common_flags_t;
endpackage : foo_flags

package fooes;

  typedef enum logic [1:0]{
    a,
    b,
    c,
    d
  } classes;

endpackage : fooes

typedef struct packed {
  logic a;
  fooes::classes b;
} padded_fooes_t;

typedef union packed {
  foo_flags::common_flags_t  [3:0][7:0] atype_t;
  padded_fooes_t     [3:0][7:0] btype_t;
} top_flag_t;

module top(input top_flag_t a , output top_flag_t b);
  assign b = a;
endmodule

```

@kamilrakoczy Can you take a look for the Google related work.

cc @hzeller

kamilrakoczy, 2023-05-04:
``unsupported_typespec`` comes from Surelog, so I've created issue there: https://github.com/chipsalliance/Surelog/issues/3630.

When ``top_flag_t`` is declared in package instead of file level, Surelog reports it correctly, so it might be temporary workaround.

kamilrakoczy, 2023-05-10:
Original issue (``unsupported_typespec``) is now fixed with newest Surelog, but now this test fails with different union members size that is tracked in: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/503.

I've also created PR that adds this test to ``UHDM-integration-tests``: https://github.com/chipsalliance/UHDM-integration-tests/pull/731

kamilrakoczy, 2023-05-16:
Surelog added missing ranges when union is defined in package, but it still lacks information about them, when union is defined on file level. I've created issue in Surelog to fix this: https://github.com/chipsalliance/Surelog/issues/3654

kamilrakoczy, 2023-05-17:
This issue should also be fixed when using submodules after: https://github.com/antmicro/yosys-systemverilog/pull/1721

@QuantamHD @grotival could you please check if this works for you?

grotival, 2023-05-23:
verified: everything is good. Thanks for the fix!