chipsalliance/yosys-f4pga-plugins #516: Handling of "'1" expression

grotival, 2023-05-16:
from https://github.com/chipsalliance/Surelog/issues/3375

Reading this Verilog with UHDM then Yosys has the wrong values.

```
typedef struct packed {logic [3:0] a;} my_struct_packed_t;

module unsized_single_bit_1 (
    output wire [3:0] out1,
    output wire out2,
    output my_struct_packed_t out3,
    output wire out4
);

  // out1 should be 4'b1111 --> OK
  assign out1   = '1;

  // out2 should be 1'b1 --> OK
  assign out2   = (out1 == 4'b1111);

  // out3 should be 4'b1111 --> NOT OK
  assign out3.a = '1;

  // out4 should be 1'b1 --> OK
  assign out4   = (out3 == '1);

endmodule : unsized_single_bit_1
```

grotival, 2023-05-16:
FYI @QuantamHD @hzeller

kamilrakoczy, 2023-05-18:
@grotival This should be fixed when using yosys-f4pga-plugins after: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/519. Could you please confirm?

grotival, 2023-05-22:
All good, thanks!