chipsalliance/yosys-f4pga-plugins #517: ERROR: Currently support for dot-access is limited to single range

grotival, 2023-05-17:
I hit this case when parsing some advanced Google SystemVerilog with UHDM + Yosys. Not sure what is the exact repro case. I can look into it more if needed, just thought I would flag it ASAP.

line throwing this error: https://github.com/chipsalliance/yosys-f4pga-plugins/blob/474724189715dd0d4fbb51d2f1cb2fa22f205722/systemverilog-plugin/UhdmAst.cc#L765

FYI: @hzeller @QuantamHD

grotival, 2023-05-17:
repro:

```systemverilog
typedef struct packed {logic [3:0][7:0] k;} my_struct_packed_t;
typedef logic [3:0][7:0] my_struct_packed2_t;

module single_range_dot_access (
    input  my_struct_packed_t x1,
    output my_struct_packed_t y1,
    input  my_struct_packed_t x2,
    output my_struct_packed_t y2
);
  assign y1.k[1][1] = x1.k[2][3];
  assign y2.k[1][1] = x2.k[2][3];
endmodule : single_range_dot_access
```

wsipak, 2023-05-19:
Thank you for the test case.
We've confirmed that there is sufficient information in the tree in order to process the multi-range access.
We're working on replacing the single range that is currently processed in the plugin to a vector of ranges that will allow calculating proper access to the selected data.

wsipak, 2023-05-23:
The RHS of the assignment lacks information in the UHDM tree, which has been reported here: https://github.com/chipsalliance/Surelog/issues/3669

wsipak, 2023-05-25:
Here are the changes needed to process LHS of the assignments, in review:
https://github.com/chipsalliance/yosys-f4pga-plugins/pull/520

wsipak, 2023-05-29:
The issue in Surelog (https://github.com/chipsalliance/Surelog/issues/3669) has been resolved, and the information about the RHS of the assignment is in the UHDM tree as of Surelog version `9bbf5427`.
However, when bumping Surelog in our repository with tests, we've observed further problems, and reported another issue in Surelog, which has also been resolved: https://github.com/chipsalliance/Surelog/issues/3675

We still have some tests failing due to the recent changes in Surelog and we're working on adjusting the plugin accordingly.

wsipak, 2023-05-31:
https://github.com/chipsalliance/yosys-f4pga-plugins/pull/520 has been merged.

wsipak, 2023-06-02:
We've reported further issues in Surelog https://github.com/chipsalliance/Surelog/issues/3692, https://github.com/chipsalliance/Surelog/issues/3691, that are related to integrating the plugin with the recent version of Surelog.

wsipak, 2023-06-06:
Another issue in Surelog has been reported [here](https://github.com/chipsalliance/Surelog/issues/3698), and in order to integrate the plugin with the new Surelog, I've prepared this: https://github.com/chipsalliance/yosys-f4pga-plugins/pull/528

wsipak, 2023-06-09:
https://github.com/chipsalliance/yosys-f4pga-plugins/pull/528 has been merged.

grotival, 2023-08-10:
Looking at https://github.com/chipsalliance/Surelog/issues/3709, it seems like this is good now?

mglb, 2023-08-19:
It is not. https://github.com/chipsalliance/yosys-f4pga-plugins/pull/521 fixed this partially, but we also need https://github.com/chipsalliance/yosys-f4pga-plugins/pull/533

mglb, 2023-08-29:
To be clear - the issue from the PR description is fixed. Following snippet works:
```systemverilog
typedef struct packed {logic [3:0][7:0] k;} my_struct_packed_t;
typedef logic [3:0][7:0] my_struct_packed2_t;

module single_range_dot_access (
    input  my_struct_packed_t x1,
    output my_struct_packed_t y1,
    input  my_struct_packed_t x2,
    output my_struct_packed_t y2
);
  assign y1.k[1][1] = x1.k[2][3];
  // assign y2[1][1] = x2[2][3];
endmodule : single_range_dot_access
```
the commented line does not work yet.

grotival, 2023-10-13:
Any update here? 
Also what would be needed to get the case `assign y2[1][1] = x2[2][3];` fixed too?