chipsalliance/yosys-f4pga-plugins #535: incorrect arithmetic shift

grotival, 2023-06-21:
repro case inspired by https://electronics.stackexchange.com/questions/132773/difference-between-and-in-verilog

Parsing the following snippet
```systemverilog
module binary_arithmetic_shift (
    output logic [4:0] b,
    output logic [4:0] c,
    output logic [4:0] d,
    output logic [4:0] e
);

  logic [4:0] a;
  assign a = 5'b10100;
  assign b = a <<< 2;  //b == 5'b10000
  assign c = a >>> 2;  //c == 5'b11101, because sign bit was `1`
  assign d = a << 2;  //d == 5'b10000
  assign e = a >> 2;  //e == 5'b00101

endmodule : binary_arithmetic_shift
```

via UHDM then reading in Yosys and then writing verilog gives the following result
```systemverilog
/* Generated by 0.28+6 */

module binary_arithmetic_shift(b, c, d, e);
  wire [4:0] a;
  output [4:0] b;
  wire [4:0] b;
  output [4:0] c;
  wire [4:0] c;
  output [4:0] d;
  wire [4:0] d;
  output [4:0] e;
  wire [4:0] e;
  assign a = 5'b10100;
  assign b = 5'b10000;  // <------ Sadness :'( 
  assign c = 5'b00101;
  assign d = 5'b10000;
  assign e = 5'b00101;
endmodule

```

grotival, 2023-06-21:
FYI @QuantamHD @hzeller

QuantamHD, 2023-06-21:
It looks like b is correct did you mean to put that comment on c?

grotival, 2023-06-21:
Yes, corrected on the above message to avoid confusion. Thanks!

mglb, 2023-07-17:
Are you sure this is a bug? `logic` is unsigned by default, according to IEEE 1800-2017, 6.11 Integer data types, Table 6-8:

> logic - 4-state data type, user-defined vector size, unsigned

The operator is expected to fill bits with zeros in such situation according to IEEE 1800-2017, 11.4.10 Shift operators:

> The arithmetic right shift **shall fill the vacated bit positions with zeros if the result type is unsigned**. It shall
> fill the vacated bit positions with the value of the most significant (i.e., sign) bit of the left operand if the
> result type is signed.

After changing all `logic` to `logic signed` in the snippet I got `c = 5'b11101`.

alaindargelas, 2023-08-29:
The tool produces the correct result. The expectation is incorrect. This issue can be closed.
I converted the design to use parameters so the answer is the same with all commercial simulators else some produce xxxx with the setup below:

```
module binary_arithmetic_shift (
);

  parameter logic [4:0] a = 5'b10100;
  
  parameter logic [4:0] b = a <<< 2;  //b == 5'b10000
   parameter logic [4:0] c = a >>> 2;  //c == 5'b11101, because sign bit was `1`
   parameter logic [4:0] d = a << 2;  //d == 5'b10000
  parameter logic [4:0] e = a >> 2;  //e == 5'b00101

  initial $display("%b", c);  
  
endmodule : binary_arithmetic_shift
```

Produces on all commercial simulators:

00101

mglb, 2023-08-29:
Thanks for confirmation, I'm closing the issue.

closed 2023-08-29 by mglb