chipsalliance/yosys-f4pga-plugins #539: Array Parameter Not Accepted

QuantamHD, 2023-08-20:
```
module test #(
  parameter int BobaCount = 2,
  parameter int unsigned NumBobaDrinkers[BobaCount] = '{default: '1}
)();
endmodule
```

Slang appears to parse this correct

```json
{
            "name": "NumBobaDrinkers",
            "kind": "Parameter",
            "addr": 6295484821520,
            "type": "int unsigned$[0:1]",
            "initializer": {
              "kind": "StructuredAssignmentPattern",
              "type": "int unsigned$[0:1]",
              "defaultSetter": {
                "kind": "UnbasedUnsizedIntegerLiteral",
                "type": "bit[0:0]",
                "value": "1'b1",
                "constant": "1'b1"
              },
              "constant": "[32'd4294967295,32'd4294967295]"
            },
            "value": "[32'd4294967295,32'd4294967295]",
            "isLocal": false,
            "isPort": true,
            "isBody": false
          }
```

QuantamHD, 2023-08-21:
@kamilrakoczy

mglb, 2023-08-22:
Seems to be issue in Surelog. Reported: https://github.com/chipsalliance/Surelog/issues/3803

alaindargelas, 2023-08-25:
https://github.com/chipsalliance/Surelog/issues/3803 is fixed.
Please propagate the fix

alaindargelas, 2023-08-30:
@mglb , please confirm you can proceed here.

tgorochowik, 2023-09-01:
Hi @alaindargelas , we're looking into this

wsipak, 2023-09-01:
As can be seen in https://github.com/chipsalliance/Surelog/issues/3803
the UHDM printer by default doesn't show all the details, and the `array_typespec` looks empty while it acutally references a proper typespec.

I've used `surelog -sv -d uhdm -d vpi_ids -parse test.sv`.

I can see that the typespec refers to `id:12`:

```
  |vpiParamAssign:
  \_param_assign: , id:102, line:3:26, endln:3:69
    |vpiParent:
    \_module_inst: work@test (work@test), id:106, file:/home/user/YOSYS/systemverilog-plugin/test.sv, line:1:1, endln:5:10
    |vpiRhs:
    \_array_expr: , id:98
      |vpiParent:
      \_param_assign: , id:102, line:3:26, endln:3:69
      |vpiTypespec:
      \_array_typespec: , id:12, line:3:13, endln:3:52
      |vpiExpr:
      \_constant: , id:92, line:3:66, endln:3:68
        |vpiDecompile:4294967295
        |vpiSize:32
        |UINT:4294967295
        |vpiConstType:9
      |vpiExpr:
      \_constant: , id:92, line:3:66, endln:3:68
    |vpiLhs:
    \_parameter: (work@test.NumBobaDrinkers), id:109, line:3:26, endln:3:41
```

I've searched for `id:12` and found this:
```
    |vpiTypespec:
    \_array_typespec: , id:12, line:3:13, endln:3:52
      |vpiParent:
      \_parameter: (work@test.NumBobaDrinkers), id:6, line:3:26, endln:3:41
      |vpiRange:
      \_range: , id:7, line:3:42, endln:3:51
        |vpiParent:
        \_parameter: (work@test.NumBobaDrinkers), id:6, line:3:26, endln:3:41
        |vpiLeftRange:
        \_constant: , id:8
          |vpiParent:
          \_range: , id:7, line:3:42, endln:3:51
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
        |vpiRightRange:
        \_operation: , id:10, line:3:42, endln:3:51
          |vpiParent:
          \_range: , id:7, line:3:42, endln:3:51
          |vpiOpType:11
          |vpiOperand:
          \_ref_obj: (work@test.NumBobaDrinkers.BobaCount), id:9, line:3:42, endln:3:51
            |vpiParent:
            \_operation: , id:10, line:3:42, endln:3:51
            |vpiName:BobaCount
            |vpiFullName:work@test.NumBobaDrinkers.BobaCount
          |vpiOperand:
          \_constant: , id:11
            |vpiSize:64
            |INT:1
            |vpiConstType:7
      |vpiElemTypespec:
      \_int_typespec: , id:5, line:3:13, endln:3:25
```

To me it looks as if the problem is now solved in Surelog, and we're going to update the Surelog submodule in yosys-systemverilog to integrate the changes.