chipsalliance/yosys-f4pga-plugins #542: latest yosys: ast process_format_str() changed, plugin won't compile.

hzeller, 2023-08-31:
Looks like the new function is now called processFormat(). So the systemverilog plug--in will not compile with latest yosys from head as our local copy of `simplify.cc` uses that in [two](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/128ed9ce9cafc1a95b26f29a98c76c1e520451a6/systemverilog-plugin/third_party/yosys/simplify.cc#L728) [places](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/128ed9ce9cafc1a95b26f29a98c76c1e520451a6/systemverilog-plugin/third_party/yosys/simplify.cc#L3334). So probably some `#ifdef` of sorts needed to work with older and newer yosys-es.

wsipak, 2023-09-01:
Thank you. We'll look into the details. Indeed it looks like `#ifdef`s are the way to go here.

hzeller, 2023-09-12:
I'll create a duplicate in the new location.

tcal-x, 2023-12-12:
Perhaps related to this, the latest plugins package doesn't work with the latest Yosys package (conda / litex-hub).   We get a runtime error with the `xdc` and `params` plugins:
```
% plugin -i params
ERROR: Can't load module `./params': /home/tim/CFU-Playground/env/conda/envs/cfu-symbiflow/bin/../share/yosys/plugins/params.so: undefined symbol: _ZN5Yosys7stringfB5cxx11EPKcz
Yosys command produced an error
```
```
% plugin -i xdc
ERROR: Can't load module `./xdc': /home/tim/CFU-Playground/env/conda/envs/cfu-symbiflow/bin/../share/yosys/plugins/xdc.so: undefined symbol: _ZN5Yosys7stringfB5cxx11EPKcz
Yosys command produced an error
```
If I pin Yosys back at 0.33, it works (the latest is 0.36).   I haven't determined precisely where this issue comes up.

hzeller, 2023-12-12:
Yep, this should work in the new synlig plugin. Don't know who is maintaining the litex hub currently.

On NixOS/nix package manager, I have a relatively recent synlig and surelog if that helps

```
nix-shell -p surelog yosys yosys-synlig

surelog --version
VERSION: 1.80

yosys --version
Yosys 0.36 (git sha1 8f07a0d8404, gcc 12.3.0 -fPIC -Os)

yosys -p "plugin -i systemverilog; read_systemverilog foo.sv"
```